tetrabiodistributed/papr-print-files #25: Respirator without hose exploded diagram

mmroden, 2021-01-27:
Need a diagram without the hose exploded diagram

mmroden, 2021-01-27:
https://tetrabiodistributed.github.io/project-tetra-docs/papra-build-guide/components/building-tetra-papr-fan-box-v1/

mmroden, 2021-01-27:
https://tetrabiodistributed.github.io/project-tetra-docs/papra-build-guide/components/building-tetra-papr-mask-v1/  but without the hose, with the pressure lock

mmroden, 2021-02-17:
fusion file created with all the components, Darragh hasn't started with the exploded diagram yet

darraghbr, 2021-02-17:
Made a start, need to get access to the PAPRA part register to assign some part numbers in order to continue. @mechatroniks-git 
[D-TET-0-0 Mask Assembly Drawing.pdf](https://github.com/tetrabiodistributed/papr-print-files/files/5998623/D-TET-0-0.Mask.Assembly.Drawing.pdf)

mmroden, 2021-02-17:
The mount goes into the mask; the mask has to be deformed to get the mount onto it.  Like so: [https://photos.smugmug.com/Tetra-Testing/Tetra-PAPRa-Build-Party-31-Jan-2021/i-VdMFNk7/0/1fce67c4/X2/_MMR0595-X2.jpg](image)

mmroden, 2021-02-17:
note also that there are tabs for the mount to fit onto the mask itself

darraghbr, 2021-02-20:
[D-TET-001104-0 Mask Assembly Drawing.pdf](https://github.com/tetrabiodistributed/papr-print-files/files/6016745/D-TET-001104-0.Mask.Assembly.Drawing.pdf)

mmroden, 2021-02-20:
You're going to hate me, but I forgot to say that we need some filter material in there.  It goes between the gasket (3) and the front element (4) (right, @DHHwangPhD ?).  We've been using material cut from 3-layer surgical masks, which still provides N95 levels of protection

darraghbr, 2021-02-20:
No worries, updated here

[D-TET-001104-0 Mask Assembly Drawing.pdf](https://github.com/tetrabiodistributed/papr-print-files/files/6016808/D-TET-001104-0.Mask.Assembly.Drawing.pdf)

mmroden, 2021-02-20:
lgtm

closed 2021-02-20 by mmroden