tetrabiodistributed/papr-print-files #27: Exploded diagram for the fan box for the Jan 2021 version of the PAPRa

mmroden, 2021-01-27:
Need an exploded diagram of the fan box

mmroden, 2021-01-27:
https://tetrabiodistributed.github.io/project-tetra-docs/papra-build-guide/components/building-tetra-papr-fan-box-v1/

darraghbr, 2021-02-10:
@mechatroniks-git @DHHwangPhD 

[image]

When I tried to add the fan box PCB I got some clipping issues when I put it where I expect it to go. I am using the file in git called TypeA_PAPR2.

mmroden, 2021-02-11:
the fan is upside down in this rendering, needs to be fitted

darraghbr, 2021-02-11:
Turns out I was looking at the wrong version. Version to use is V4.2.

darraghbr, 2021-02-17:
I've generated a milestone in Fusion. In Tetra PAPR\Archive\PAPRA V4\SinglePAPRv4 I've labelled it 'January version' is this the correct version of the assembly that was made in January? @DHHwangPhD  @mmroden

mmroden, 2021-02-17:
that's the one from Jan 6th, so it could do in a pinch.  The next version used these parts: 
[image], which looks like it might be model 4.1 with the most recent top plate and a base plate missing

darraghbr, 2021-02-17:
Are there still supports on the fan box outlet? I can't find a part that looks like that in any of the versions.

mmroden, 2021-02-17:
Check the funnel in the part you mentioned: Tetra PAPR\Archive\PAPRA V4\SinglePAPRv4

[image]

That would require supports

darraghbr, 2021-02-17:
Yep that's all fine, I'm just trying to make sure I use the right version.

mmroden, 2021-02-17:
gotcha.  I think the one we built had a part number on it, so it's not the one in that diagram; I think it's the same thing but with a part number.

mmroden, 2021-02-24:
not really necessary for the milestone

darraghbr, 2021-03-04:
solved in #82 ?

mmroden, 2021-03-12:
No bandwidth, won't do

closed 2021-03-12 by mmroden