tetrabiodistributed/papr-print-files #29: exploded diagram of the v2 of the m12 battery controller

mmroden, 2021-01-27:
For these instructions:
https://tetrabiodistributed.github.io/project-tetra-docs/papra-build-guide/components/building-tetra-papr-m12-v2/

mechatroniks-git, 2021-01-27:
Is this along the lines of what you want?
[image]

mmroden, 2021-01-27:
That would be great!  If we had the ability to label the various bits, that
would be an extra awesome step, but I realize that may be a bridge too far

mechatroniks-git, 2021-01-27:
Or this? And image or pdf?

[image]

mmroden, 2021-01-27:
The second one is better; are the nuts not labeled because they are part of another part?

Image is great, pdf is a great addition, but image is the main deliverable I think.  PNG ideally, no jpg.

mechatroniks-git, 2021-01-27:
Ok, the parts are grouped in assemblies, so you only get 1 number for the assembly, might be a way to do multi-level as a flat assembly, but will get all the board level parts. In any case, there is lots of cleanup to do on the files to have a part list that is correct. This will take some time.

mmroden, 2021-01-27:
We probably need to do this for proper manufacturing handoff, I'd think.

mechatroniks-git, 2021-01-28:
Yes, lots of busy work getting this done right. Assigning part numbers, updating each part's material property and part #, then the part's list will be good for exploded view drawings.

darraghbr, 2021-02-10:
[A-TET-001102-0 M12 PAPRA Controller Drawing.pdf](https://github.com/tetrabiodistributed/papr-print-files/files/5961147/A-TET-001102-0.M12.PAPRA.Controller.Drawing.pdf)

Should be good to go @mmroden

mmroden, 2021-02-12:
Perfect, got the fixed lid in there too!

closed 2021-02-12 by mmroden