tetrabiodistributed/papr-print-files #58: part registry

mmroden, 2021-02-11:
need a central part registry for all tetra parts
should be its own repo
comfortable for just being a google sheet
change history for adding new parts (should be a google sheet)
should have proper access control

P-TET exists
A-TET assemblies
D-TET drawings

Should be its own thing, just recording this hear

mechatroniks-git, 2021-02-11:
Currently we've been using:

```
A-TET-ZZZZZZ-R = Assembly 
    ZZZZZZ-R is Assembly Number & Revision
P-TET-ZZZZZZ-R = Part
    ZZZZZZ-R is Part Number & Revision
M-YYY-ZZZZZ = Buy Part, M = Manufacturer, YYY = manufacturer code (ex, UNI = Unistrut)
    ZZZZZ = Manufacturer part number
V-XXX-ZZZZZ = OTS part, V = Vendor, XXX = vendor code (ex, MCM = McMaster-Carr)
    ZZZZ = Vendor part number 
```

darraghbr, 2021-02-11:
So I envision the solution to this being a form.

This form should have 4 tick boxes to specify whether you want to request a part, assembly, drawing or document number.

A field to accept a string with the name of the part or assembly or drawing etc.

Another field to accept a string with the name of the requester. 

Upon submission the form should return the next unused part, assembly, drawing etc number.

mechatroniks-git, 2021-02-11:
Currently found part #s here: 
https://docs.google.com/spreadsheets/d/1nCw9LUN7ygv8B8hno6poN-m15rsixNSolVTXhXynrfw/edit?pli=1#gid=0
https://docs.google.com/spreadsheets/d/1SvVoTHTWQuhzA31efnyqhaFsMOlKPyU90ATQrHQNn1k/edit#gid=0

darraghbr, 2021-02-13:
I have prototyped a system to issue numbers using something called Appsheet that uses a google sheet in the background, needs to be paid if we want to deploy it. I am investigating funding at the moment. 

This solution is pretty quick and dirty, I think that if multiple people were requesting the same thing at the same time it would probably break a bit.

I think we should also issue document numbers that can then get thrown into a document register. What designation should we use for documents?

Currently I have P-TET for parts, A-TET for assemblies, D-TET for drawings.

capsulecorplab, 2021-02-14:
I think this issue falls under a bigger discussion of how we internally facilitate product lifecycle management (PLM) for an open-source hardware (OSHW) project. I would be cautious of jumping the gun on using a service that requires a paywall, as those generally tend to contradict the principles of OSHW.

darraghbr, 2021-02-14:
It would be absolutely nothing more than a stop gap measure. 

While the service does require funding the only way we would lose the data is if the google sheet (which is currently in my personal account but can be moved) were lost for some reason.

mmroden, 2021-02-20:
@mechatroniks-git created a new repo for part numbers: https://github.com/tetrabiodistributed/tetraPLM

closed 2021-02-20 by mmroden