tetrabiodistributed/papr-print-files #59: Document different running and crossfit workout belts

mmroden, 2021-02-12:
Testing different belts (molle, runner's belt) with running and crossfit

mmroden, 2021-02-12:
Diana will be doing this work, Mark will record it

mmroden, 2021-02-20:
Diana has a request for updates to the base plate
[image]
[image]
[image]

mmroden, 2021-02-24:
More updates with jumping

[image]
[image]

https://user-images.githubusercontent.com/116251/109040212-87bbf400-7682-11eb-970f-068b328ca8c0.mp4

DHHwangPhD, 2021-03-05:
Lacing of webbing for securing the PAPR using the new MOLLE Adapter Short plates as well as securing the M12 controller.

[image]
[image]
[image]
[image]
[image]

mmroden, 2021-03-26:
Documentation is now in the march build

closed 2021-03-26 by mmroden