tetrabiodistributed/papr-print-files #62: Create A4/Letter versions of the gaskets

mmroden, 2021-02-15:
There is currently only an A3 template for the gaskets

darraghbr, 2021-02-17:
Made a letter version, the question I have is should I resize the drawing box so that it all fits on one row or is what I have done fine? @mmroden 

[Tetra Papr Drawing Template A v2.pdf](https://github.com/tetrabiodistributed/papr-print-files/files/5998632/Tetra.Papr.Drawing.Template.A.v2.pdf)

mmroden, 2021-02-17:
What do you mean by "it all fits on one row"?  I'm not sure the ramifications of that decision

darraghbr, 2021-02-17:
Well if you look at the pdf I've uploaded, the section with the 3rd angle projection symbol, the tolerances and units are in their own little box above the main drawing box, whereas in the D size version all of that info is in one box at the bottom right of the page. 

It's purely a stylistic thing and if you don't mind the way it looks I will continue.

mmroden, 2021-02-17:
I do not mind the way it looks at all.

The way @DHHwangPhD had his gaskets cut by his cricut, he had them one inside of the other to save material-- can that be reflected here?  Or is that a different kind of concern?

darraghbr, 2021-02-17:
If we are also uploading .dxf files for all of the gaskets that is up to the end user. It is entirely possibly to nest them inside one another.

mmroden, 2021-02-17:
works for me, we are definitely uploading dxf files to the gaskets folder here

darraghbr, 2021-02-22:
I've finished the letter sized versions of these drawings.

mmroden, 2021-02-22:
Maybe a PR if they're not in the appropriate directory?

darraghbr, 2021-02-22:
They're in #69

mmroden, 2021-03-03:
Merged

closed 2021-03-03 by mmroden