tetrabiodistributed/papr-print-files #63: Exploded diagram for the Feb 2021 fan box

mmroden, 2021-02-17:
now with the gasket for the funnel instead of the snap to fit

darraghbr, 2021-02-17:
I tried to get the view from the other side but to no avail. I need an assembly number for this set up as well @mechatroniks-git 

[D-TET-0-0 Fan Box Drawing.pdf](https://github.com/tetrabiodistributed/papr-print-files/files/5998568/D-TET-0-0.Fan.Box.Drawing.pdf)

mechatroniks-git, 2021-02-20:
Updates to parts list:
[image]

I prefer using tangent edges, shortened on the view
[image]

darraghbr, 2021-02-20:
Thanks Patrick, I have addressed your comments and given it an assembly number in this version.

[D-TET-001103-0 Fan Box Drawing.pdf](https://github.com/tetrabiodistributed/papr-print-files/files/6016193/D-TET-001103-0.Fan.Box.Drawing.pdf)

mmroden, 2021-02-20:
Hole placement for the new fan controller board will likely need to be updated, but!  This is sufficient for assembly

closed 2021-02-20 by mmroden