tetrabiodistributed/papr-print-files #70: Generate ISO standard drawing templates

darraghbr, 2021-02-22:
I've generated some drawing templates that fit the ISO standard so we can issue drawings under that standard.

@mmroden please review when able.

[Tetra A4 Drawing Template.pdf](https://github.com/tetrabiodistributed/papr-print-files/files/6025045/Tetra.A4.Drawing.Template.pdf)
[Tetra A3 Drawing Template.pdf](https://github.com/tetrabiodistributed/papr-print-files/files/6025046/Tetra.A3.Drawing.Template.pdf)

mmroden, 2021-02-22:
These LGTM, but I would ask @mechatroniks-git what he thinks, since this is my first real rodeo with mechanical drawings

mechatroniks-git, 2021-02-22:
Looks good. We had planned on making templates for assembly and part drawings. With an additional part drawing template specifically for 3D printed parts. Not sure we need all those, but the assembly comment makes the template destined for assembly drawings.

closed 2021-03-01 by capsulecorplab