tetrabiodistributed/papr-print-files #71: Validate that fan box STLs have been properly constructed

mmroden, 2021-02-24:
The Fan Box STLs have not been printing well on a prusa 3d.  @darraghbr thinks that there might be fan box issues

mmroden, 2021-02-24:
[image]

mmroden, 2021-02-24:
[P-TET-000112-2 Fan Box_0.2mm_PLA_MK3S_8h12m.gcode.zip](https://github.com/tetrabiodistributed/papr-print-files/files/6037672/P-TET-000112-2.Fan.Box_0.2mm_PLA_MK3S_8h12m.gcode.zip)

mmroden, 2021-02-24:
[P-TET-000112-2 Fan Box.3mf.zip](https://github.com/tetrabiodistributed/papr-print-files/files/6037679/P-TET-000112-2.Fan.Box.3mf.zip)

Judoguy12, 2021-02-25:
Prints fine on my machine 
[image]

Judoguy12, 2021-02-25:
[image]

mmroden, 2021-02-26:
Jamie can print it, I have problems

closed 2021-02-26 by mmroden