tetrabiodistributed/papr-print-files #76: Loose m12 controller lid

capsulecorplab, 2021-02-26:
P-TET-000102-1 (a.k.a., m12 controller lid) seemed to be a loose fit when placed into the controller body
[image]
[image]

mechatroniks-git, 2021-03-01:
@capsulecorplab which revision of P-TET-000104 M12 Controller Body.stl did you print & test?

mechatroniks-git, 2021-03-01:
I'll print P-TET-000102-1 and P-TET-000104-3 and test.

mechatroniks-git, 2021-03-02:
[image]

Lid is held with friction only. It pops out easily when tapping the body. A positive lock would be better, perhaps the tabs can be longer and fit into grooves in the body?

capsulecorplab, 2021-03-04:
@mechatroniks-git looks like you found it, but for the record, it was https://github.com/tetrabiodistributed/papr-print-files/commit/fd07e96a460faa9d1c7b4d5da6ee4366efcfc87a

closed 2021-05-13 by mmroden