tetrabiodistributed/papr-print-files #96: Fan box jack issues

DHHwangPhD, 2021-03-15:
Problems with the power jack in the fan box.
Original power jack for the board is not sealed.
New power jack is sealed, but will require custom cables.
Other jacks like M12 have been discussed with @mechatroniks-git

darraghbr, 2021-03-15:
Can you expand on why you think the new connector needs a custom cable? The jack is for a standard 5.5 mm x 2 mm plug.

mechatroniks-git, 2021-03-15:
The connector entry is deeper and has an oring seal, so the cable end needs to be longer and match the diameter at the seal

mmroden, 2021-03-15:
yeah @darraghbr-- @mechatroniks-git had to sand down a connector to make a custom connector for a cable.  So either we need another jack or we need custom cables.

darraghbr, 2021-03-15:
By the looks of things the compatible jack uses a thread to pull the 2 parts together to create a seal.

https://www.mouser.co.uk/ProductDetail/Switchcraft/S761KS15?qs=fbkhFuCHoTam5cvHIFHKsw%3D%3D

mechatroniks-git, 2021-03-19:
[image]
BNC Cable: https://www.monoprice.com/product?p_id=16180
BNC Connector: https://www.digikey.com/en/products/detail/adam-tech/RF1-01A-D-00-75-M/9831054

mmroden, 2021-03-20:
Will likely close if the BNC connector route works: #102

closed 2021-05-13 by mmroden