tetrabiodistributed/papr-print-files #105: add louvers to the filter cover

mmroden, 2021-03-20:
must determine the direction before designing can start, likely a conversation about how to mount the devices.

Judoguy12, 2021-04-09:
I  have tested this, for light rain the louvres are too small to prevent water hitting the filter, they need to be much larger to be effective. They printed nicely though

darraghbr, 2021-04-09:
There are a couple of ways that you can take this, you can either go mushroom cowl style, where you have a curved section that stands off from the body of the PAPRA and allows air to come in through a circuitous path. Something similar to below. I would suggest that the 'open' area is nominally similar to the intake area of the fan we are using. If you wanted to you could make the cowl draw in air only from the bottom and sides of the PAPRA, leaving the top completely closed. 

[image]

The other way you can do this is more conventional louvers. A profile like shown below is pretty common. You could use some solid sections at points across the 'span' to aid in printing and in ridigity of the final component. 

[image]

closed 2021-05-13 by mmroden