tetrabiodistributed/papr-print-files #115: Black wall on M12 controller at 2 layers blocks LED

mechatroniks-git, 2021-04-10:
Current limiting resistor for LED swapped out to 100 ohms, LED is driven at 18mA

[image]

mmroden, 2021-04-12:
Not a problem for the pink papr

mmroden, 2021-04-12:
not a problem for grey-- blue may not shine through black very well?  Maybe spec a brighter LED?

mmroden, 2021-04-12:
we do need to spec the specific LED.  Don't have a problem with superbrights, because have to go through plastic.  Didn't matter with direct exposure with the light pipe, but now that the light pipe is encased in plastic, should use superbrights

mmroden, 2021-04-12:
targeting a red LED

mmroden, 2021-04-12:
red LEDs are hard to get right now, may have to look at a different color.

mechatroniks-git, 2021-04-12:
Blue LED: https://www.we-online.de/katalog/datasheet/150080BS75000.pdf
With 100ohm LEDs, we were driving the led at about 18mA, which is about 135mcd. These LEDs have a 140° viewing angle, which is close to the 160° recommended [here](https://vcclite.com/light-pipe-faqs/#:~:text=For%20ideal%20illumination%20whether%20using,viewing%20angle%20of%20160%20degrees.). 

[image]

mechatroniks-git, 2021-04-12:
Recommend using this [JLC LED](https://lcsc.com/product-detail/Light-Emitting-Diodes-LED_Foshan-NationStar-Optoelectronics-NCD0603G1_C84267.html) see [datasheet](https://datasheet.lcsc.com/szlcsc/2008201032_Foshan-NationStar-Optoelectronics-NCD0603G1_C84267.pdf)

mechatroniks-git, 2021-04-24:
[image]

mmroden, 2021-05-13:
Don't print it in black; make sure to document

closed 2021-07-17 by mmroden