tetrabiodistributed/papr-print-files #127: PETG prints still not perfect

mechatroniks-git, 2021-05-09:
Seeing issues on the longer parts:
Ender 3 Pro  with blobs/boogers
Prusa has ripples on the inside
Makerbot M12 controllers were fine

argus1, 2021-05-13:
trying transparent PETG

mechatroniks-git, 2021-05-16:
I printed a temperature tower, found that 235'C worked well for MH Build PETG on Replicator 2
Simplify3D settings:
- Bed temp at 80'C (layer 1) then 55'C (layer 2+)
- Extrusion multiplier .99
- Extrusion width .45mm (Nozzle dia .4mm)
- Retraction set to 4mm
- Retraction speed 1800mm/min
- .3mm Coasting distance
- 1.0mm Wipe Distance
- Downward cooling fan set to 40%
- Print speed 3500 mm/min

mmroden, 2021-08-14:
@capsulecorplab can test on his ender3, he has new cura profiles

capsulecorplab, 2021-08-14:
@mechatroniks-git which parts in particular were having issues?

mechatroniks-git, 2021-08-14:
M12 controller was having print quality issues from the non-makerbot machines, as noted during build party.

capsulecorplab, 2021-08-14:
I'm currently printing at 250 with fan speed at 100%

mmroden, 2021-08-18:
Need to define what printed parts are acceptable and how we determine that acceptability