tetrabiodistributed/papr-print-files #150: Check a fan taper to see if it will work with this model fan

mmroden, 2021-06-05:
Design the taper for testing to see if we can get to the 4 CFM requirement

https://www.digikey.co.uk/product-detail/en/sunon-fans/PF97331B1-B00U-A99/259-1925-ND/6189086

or

https://www.cuidevices.com/product/thermal-management/dc-fans/centrifugal-blowers/cbm-97b-series

mechatroniks-git, 2021-06-07:
Snapshot of distributor stock:

[image]

[image]

darraghbr, 2021-06-07:
https://www.digikey.co.uk/product-detail/en/delta-electronics/BFB1012EH-A/603-2105-ND/9351771

mechatroniks-git, 2021-06-07:
[image]

mechatroniks-git, 2021-06-21:
Results from testing. Blower inlet to open air, short cone "TestFunnelShort" taped to blower with 36 inches of blue hose to Sensirion to open air. 

[image]

All 3 blowers would require a 11V minimum to meet 4CFM. The M12 battery will start at 12.4V and drain down to 8.4V, making the CUI CBM-120B series a better fit with our battery and electronics. See [prior test](https://github.com/tetrabiodistributed/papr-print-files/issues/151#issuecomment-856045012)

mmroden, 2021-07-07:
This work is done, we know we need to hit 4CFM

closed 2021-07-07 by mmroden