tetrabiodistributed/papr-print-files #151: Test the fan taper to ensure that 4 CFM is hit

mmroden, 2021-06-05:
blocked on #150 ; once we have a fan taper, we can then redesign the box for the fan

mechatroniks-git, 2021-06-07:
**Airflow tests with OLD funnel**

NMB BG1203-B045-00S FAN + FUNNEL + 36" BLUE HOSE + FLOW SENSOR
12V | .44A | 52SLM

**Airflow tests with NEW funnel**

NMB BG1203-B045-00S FAN + FUNNEL + 36" BLUE HOSE + FLOW SENSOR
12V | .51A | 70SLM

UTUO FAN + FUNNEL + 36" BLUE HOSE + FLOW SENSOR
12V | .33A | 60SLM

We see improvement, but not enough to hit the target

mechatroniks-git, 2021-06-07:
Tested the CUI blower without disassembling (in fan box with HEPA filter) using the legacy funnel as attached. 

We able to get 4CFM easily at around 8V & 2A.  Could be a drop in solution to our challenges. The voltage range matches up with the M12 battery and controller electronics. 

[image]

mmroden, 2021-07-07:
This is done in the sense that we know the answer; the taper has been defined and models must be made

closed 2021-07-07 by mmroden