tetrabiodistributed/papr-print-files #158: Investigate new BNC cables

mmroden, 2021-08-18:
Need to update the BNC cables in the BOM to reflect the proper gauge for the power draw

mechatroniks-git, 2021-08-18:
Options:
20 awg - https://www.jameco.com/z/205-523-Steren-Electronics-Cable-Assembly-RG58-U-36-Bnc-To-Bnc-50-Ohm_11404.html

26 awg (rated for 2.2A)- https://www.jameco.com/z/G7402-R-Jameco-Valuepro-Cable-Assembly-Rg174-U-3-BNC-to-BNC-50-Ohm_111473.html

mmroden, 2021-08-23:
Kevin to help review

mmroden, 2021-08-28:
"V-MCM-6641T11	" will be tested

mmroden, 2021-09-13:
New BNC cable worked.

mmroden, 2021-09-13:
Apparently "plug it in and it works" isn't enough.  My bad.

mechatroniks-git, 2021-09-20:
@DHHwangPhD Which BNC did you order from McMaster?
Either one of these would work
[image]

From [Engineering Toolbox](https://www.engineeringtoolbox.com/wire-gauges-d_419.html), we're good to 5A

[image]

mmroden, 2021-11-04:
max amperage we have in current models is 1.3-1.4, so maybe target 2 amps so we have a lot of breathing room.

22 gauge should be more than fine, 24 gauge should also be OK if we can find it.

RG-59/U will work.

closed 2021-11-04 by mmroden