tetrabiodistributed/papr-print-files #220: The pressure tap model needs to get built

mmroden, 2022-10-12:
I've got a pull request set up for the pressure tap setup, but unfortunately, Prusa Slicer won't remove negative spaces.  So, I need someone to remove the negative spaces as per the attached image (ie, only about a mm thickess)

[image]

mmroden, 2022-10-15:
[Pressure Tap v3.zip](https://github.com/tetrabiodistributed/papr-print-files/files/9792716/Pressure.Tap.v3.zip)
Here is the 3mf file that is the source

Judoguy12, 2022-11-02:
I have no thin walls on my slicer although it appears to be very thin walls so would be better suited to resin

mmroden, 2022-11-02:
@DHHwangPhD has an updated model for this, would need a PR to close using that model with a new part number.  I may just do that right now

mmroden, 2022-11-03:
Closed by #224

closed 2022-11-03 by mmroden