tetrabiodistributed/papr-print-files #225: approach for keeping git commits and Fusion360 design history in sync

capsulecorplab, 2023-01-02:
as follow up to https://github.com/tetrabiodistributed/papra/issues/99, Fusion users need a means to reference git commits in Fusion360 design history.

Desired outcomes:
- export process for Fusion model bodies as STEP/STL
- process for creating a git commit directly from Fusion design history save point (including version description notes for commit message)
- process for re-configuring python environment to not break per Fusion updates

Burhan-Q, 2023-01-07:
# References:

## Fusion360 Python
[Fusion360 GitHub](https://github.com/AutodeskFusion360)
[OctoFusion](https://github.com/tapnair/OctoFusion) good reference for Add-In for Fusion, look to use for following our documented approach for file naming assistance
[tapnair F360 repos](https://github.com/tapnair)
[tapnair personal page](https://tapnair.github.io/index.html#general_anchor) Also good for reference to UI elements

## python module imports for F360
[method1](https://forums.autodesk.com/t5/fusion-360-api-and-scripts/import-data-from-excel-type-xlsx/m-p/11245348/highlight/true#M16701)
[method2](https://forums.autodesk.com/t5/fusion-360-api-and-scripts/importing-modules-error-importerror-no-module-named-openpyxl/td-p/8512051)

## python control for GitHub 
https://github.com/PyGithub/PyGithub
- request records of commits? [get_commits](https://pygithub.readthedocs.io/en/latest/github_objects/Repository.html#github.Repository.Repository.get_commits) 
- create new commit? [create_git_commit](https://pygithub.readthedocs.io/en/latest/github_objects/Repository.html#github.Repository.Repository.create_git_commit)
- blob, not sure if we need it, but blob [create_git_blob](https://pygithub.readthedocs.io/en/latest/github_objects/Repository.html#github.Repository.Repository.create_git_blob)
- maybe could use to generate text/csv file with the parts exported with the FusionVersionID numbers in case files are not uploaded [create_file](https://pygithub.readthedocs.io/en/latest/github_objects/Repository.html#github.Repository.Repository.create_file)

Burhan-Q, 2023-01-07:
@capsulecorplab will look into figuring out the GitHub controls

Burhan-Q, 2023-01-07:
[FusionDataUtils](https://github.com/tapnair/FusionDataUtils) could be helpful if looking to capture additional from Fusion